m-labs/migen #10: migen CI support

mithro, 2015-04-14:
Travis-CI has pretty good support for doing CI on python packages, see http://docs.travis-ci.com/user/languages/python/ -- It would be nice for migen to use this.

I'm guessing that might be a low priority as migen is already tested by misoc?

fallen, 2015-04-14:
Indeed Migen is already somehow tested by MiSoC builds ... BUT, it's also very interesting to have Travis-CI for Migen, because it then allows to automate generation and upload of conda packages for Migen, as it is done for ARTIQ.

mithro, 2015-04-14:
I sent some patches to the mailing list to enable this. See
https://ssl.serverraum.org/lists-archive/devel/2015-April/003987.html

closed 2015-04-14 by sbourdeauducq