m-labs/migen #12: migen version of "unknown" in setup.py is invalid

mithro, 2015-04-14:
```
/home/travis/virtualenv/python3.4.2/lib/python3.4/site-packages/setuptools/dist.py:292: UserWarning: The version specified ('unknown') is an invalid version, this may not work as expected with newer versions of setuptools, pip, and PyPI. Please see PEP 440 for more details.
```

peteut, 2015-04-17:
It should be `sys.version_info[:2]`, which works with PY3 and PY2.x. I sent a patch to the list.

sbourdeauducq, 2015-04-17:
I think the issue refers to this: https://github.com/m-labs/migen/blob/master/setup.py#L18

peteut, 2015-04-17:
Oh, I see. This indeed violates PEP 440. I use `pbr` for my own project to be freed of those kind of issues. 
`pbr` generates a valid version number from the git commit. Any idea how to fix this?

Best regards,
Alain

closed 2015-05-28 by sbourdeauducq