m-labs/migen #13: migen README still doesn't link to the readthedocs.org documentation - is there something wrong with it?

mithro, 2015-04-19:
Looks like when my readme-update changes were merged the following was changed;
```diff
++<<<<<<< HEAD
 +http://m-labs.hk/gateware.html
++=======
+ http://migen.readthedocs.org/en/latest/
+ 
+ Install instructions:
+ https://migen.readthedocs.org/en/latest/introduction.html#installing-migen
++>>>>>>> 62d264e... Adding links to the readthedocs.org documentation.
```

Adding the http://m-labs.hk/gateware.html link is a good idea (the user guide, tutorial and lecture slide stuff is pretty awesome!)

However, I was wondering why the links to readthedocs documentation was dropped? I was also unable to find any links to the readthedocs page on the gateware page. Is there something wrong with the documentation at migen.readthedocs.org?

sbourdeauducq, 2015-04-19:
The readthedocs content is fine, but I'd just like to keep all docs in one place (the artiq and pyparser html docs are already on m-labs.hk, as well as the pdf docs for migen).

mithro, 2015-04-20:
Could we add a link to the readthedocs stuff on http://m-labs.hk/gateware.html then?

sbourdeauducq, 2015-04-20:
What's wrong with the existing pdf?

mithro, 2015-04-20:
I didn't actually even realise it was the same documentation, I find the HTML documents much nicer to read personally.

The HTML documents are also updated on every commit to your repo if you add a github hook too.

sbourdeauducq, 2015-09-20:
Let's put all the docs in http://m-labs.hk/migen/manual/
(This is for the new branch)

closed 2015-09-20 by sbourdeauducq