m-labs/migen #16: mibuild: Support FPGALink / NeroJTAG as a programmer (and maybe computer interface?)

mithro, 2015-04-19:
NeroJTAG is a fully open specification for FPGA programmers as part of the FPGALink project (https://github.com/makestuff/libfpgalink/wiki/FPGALink) by makestuff (http://github.com/makestuff). There exists both an AVR based and Cypress FX2 based implementations of the firmware. 

It would be great if FPGALink was supported as a "programmer" type similar to xc3sprog. (It would also be awesome if there was support for using FPGALink to communicate with the FPGA board but that is probably more a misoc rather than migen thing?)

The Cypress FX2 firmware supports a large number of Digilent's boards, replacing the proprietary "Adept" firmware. (Of interest too myself is the Digilent Atlys board.) From the FPGALink wiki,
> ### Quick-starts for several devkits
> This is just a selection; there are [many more supported configurations](https://github.com/makestuff/swled/tree/20140524/templates).
> [Quickstart](https://gist.github.com/makestuff/caa2e03802e8b3a7cc3a) for [Aessent aes220](http://www.aessent.com/products/aes220-minimodule.html)
> [Quickstart](https://gist.github.com/makestuff/9392595) for [Canton Electronics LX9](http://www.canton-electronics.com/xilinx-spartan6-xc6slx9-fpga-board-vivado-ise-replace-cpld-artix-kintex-virtex-p-798.html) (requires external AVR board)
> [Quickstart](https://gist.github.com/makestuff/7486988) for [Digilent Atlys](http://www.digilentinc.com/atlys)
> [Quickstart](https://gist.github.com/makestuff/7357925) for [Digilent FMC Carrier S6](http://www.digilentinc.com/Products/Detail.cfm?Prod=FMC-CARRIER-S6)
> [Quickstart](https://gist.github.com/makestuff/7829484) for [Digilent Nexys2](http://www.digilentinc.com/nexys2)
> [Quickstart](https://gist.github.com/makestuff/9870450) for [Digilent Nexys3](http://www.digilentinc.com/nexys3)
> [Quickstart](https://gist.github.com/makestuff/51416b80db0420ad4713) for [MakeStuff LX9R3](https://github.com/makestuff/lx9/tree/r3)

enjoy-digital, 2015-04-19:
"(It would also be awesome if there was support for using FPGALink to communicate with the FPGA board but that is probably more a misoc rather than migen thing?)"

--> This is the purpose of the LiteUSB core in MiSoC. For now it only supports FT2232h chip in synchronous slave fifo mode. We also have the host software to communicate with the FPGA (DMA, virtual UARTs). Supporting the FX2 should only be a matter of adding a new PHY for it.

enjoy-digital, 2015-07-02:
Added by mithro.

closed 2015-07-02 by enjoy-digital