m-labs/migen #40: Get consistent I/O names on top level

enjoy-digital, 2016-03-04:
I/O names can change when little modifications are done to the code.

We should to to use name_override on I/Os to get consistent interfaces (useful when doing top level simulation with an external simulator or when building a core).

Here is a quick and dirty implementation:
https://github.com/enjoy-digital/litex/commit/a716c562f07611c4a0eb9ff7afd3b29f17d70bf5

Do you see a better way to do this?

sbourdeauducq, 2016-03-16:
Are you using the platform infrastructure when exporting cores?

enjoy-digital, 2016-03-16:
Yes I'm using platform to export cores (I find it convenient, here the platform is simply our core interface).

But apart from that, when generating design with Migen somes of the I/Os can be prefixed with the name of the class, some not. For example with actual MiniSoC on KC705:
````
module top(
	input minisoc_serial_cts,
	input minisoc_serial_rts,
	output reg minisoc_serial_tx,
	input minisoc_serial_rx,
	input clk200_p,
	input clk200_n,
	input cpu_reset,
	output [15:0] ddram_a,
	output [2:0] ddram_ba,
	output ddram_ras_n,
    [...]
	input eth_col,
	input eth_crs
);
````
That would be better to have serial_yy instead of minisoc_serial_yy, ie that the I/O get the name given in the platform file. This way when you want to simulate for example the entire SoC with another simulator, you don't have to change your testbench wrapper when you change something in the design. And more generally your SoC will always have the same I/Os, wich is probably better.

commit 4df34c7b (2016-03-16): fhdl/verilog: simpler names for IOs. Closes #40
closed 2016-03-16 by sbourdeauducq