m-labs/migen #41: Records with 0 fields generate invalid verilog

nakengelhardt, 2016-03-17:
When defining a record with no fields, as in 
`_layout = []`
and then using `raw_bits()`, it will generate invalid verilog, e.g.:
`assign {} = x;`

jordens, 2016-03-17:
Should it silently drop logic related to the `Record` or should it error?

nakengelhardt, 2016-03-17:
Silently drop would be the preferred solution in my case, since I use it to build generic logic where the contents of the `Record` change with the application (and some simply don't need extra data).

jordens, 2016-03-17:
Hmm. That might lead to tough questions if the Record (or its raw_bits()) is used in expressions. What's the boolean of a zero length signal? Can it be signed? How does it extend? If you drop a zero length signal from an expression and nothing is left, do you drop entire assignment? It might need to be an error.

whitequark, 2019-07-05:
Triage: fixed in nMigen, where 0-bit signals are explicitly valid.