m-labs/migen #42: stabilize and de-randomize verilog output

jordens, 2016-03-21:
https://github.com/m-labs/artiq/issues/341

mithro, 2016-03-21:
Hrm, I thought we had fixed most of this - see https://github.com/m-labs/misoc/issues/20

jordens, 2016-03-21:
Nope.

sbourdeauducq, 2016-03-21:
Fixed the non-determinism in the ARTIQ Pipistrello bitstream.

closed 2016-03-21 by sbourdeauducq

mithro, 2016-03-23:
@sbourdeauducq How did you test the non-determinism in the ARTIQ Pipistrello gateware?

sbourdeauducq, 2016-03-23:
```
#TARGET=misoc.targets.pipistrello
TARGET=artiq.gateware.targets.pipistrello

python -m $TARGET --no-compile-software --no-compile-gateware --output-dir nd1
python -m $TARGET --no-compile-software --no-compile-gateware --output-dir nd2
diff -Nau nd1/gateware/top.v nd2/gateware/top.v
```

and run in a loop.

mithro, 2016-03-23:
You might want to also check with something like this;
```bash
TARGET=artiq.gateware.targets.pipistrello

PYTHONHASHSEED=1 python -m $TARGET --no-compile-software --no-compile-gateware --output-dir nd1
PYTHONHASHSEED=2 python -m $TARGET --no-compile-software --no-compile-gateware --output-dir nd2
diff -Nau nd1/gateware/top.v nd2/gateware/top.v
```