m-labs/migen #54: support reset-less Signals

jordens, 2016-11-15:
Support for reset-less `Signals` would avoid having to create a reset_less clock domain just to be able to have a single reset_less register. E.g. for inferring shift registers in spartan 6.

sbourdeauducq, 2016-11-16:
That's not very easy to do, e.g. ``self.sync += If(foo, with_reset.eq(bar), without_reset.eq(quux))`` needs to generate two ``if``, and so on.

sbourdeauducq, 2016-11-16:
Or we can just add a ``if(reset) foo <= 0 ...`` section at the end of sync always blocks, but that's an unusual coding style and I don't know if synthesizers will recognize and optimize the reset.

jordens, 2016-11-16:
I don't see why there would be two `if`.

With e.g. `without_reset = Signal(reset_less=True)`, I'd like to see

```verilog
reg foo = 1'd0;
reg bar = 1'd0;
reg quux = 1'd0;
reg with_reset = 1'd0;
reg without_reset = 1'd0;

always @(posedge sys_clk) begin
        if (sys_rst) begin
                with_reset <= 1'd0;
        end else begin
                if (foo) begin
                        with_reset <= bar;
                        without_reset <= quux;
                end
        end
end
```

instead of 

```verilog
reg foo = 1'd0;
reg bar = 1'd0;
reg quux = 1'd0;
reg with_reset = 1'd0;
reg without_reset = 1'd0;

always @(posedge sys_clk) begin
        if (sys_rst) begin
                with_reset <= 1'd0;
                without_reset <= 1'd0;
        end else begin
                if (foo) begin
                        with_reset <= bar;
                        without_reset <= quux;
                end
        end
end
```

sbourdeauducq, 2016-11-16:
Then ~sys_rst acts as a clock enable for your "without reset" signal.

commit c1da39d4 (2017-06-18): support reset-less Signals (closes #54)

Support for reset-less Signals avoids having to create a reset_less
clock domain just to be able to have a single reset_less register.

This is helpful when inferring primitives that do not support resets,
e.g. SRL* shift registers in spartan 6.

As an important sideeffect, the synchronous reset logic has been
rewritten from
	if (rst) <rst> else <code> end
to
	<code> if (rst) <rst>
closed 2017-06-18 by sbourdeauducq