m-labs/migen #70: export as hierarchy (verilog modules)

jordens, 2017-06-19:
I'd like to be able to translate the migen Module hierarchy into the exported HDL hierarchy at least to some extent.

* This would allow estimating resource usage at some granularity.
* It would allow easier exporting and re-use of migen gateware in verilog/vhdl projects.
* I don't expect this to be able to map any of the metaprogramming and parametrization in migen to parametrization in verilog. That won't work. I am fine with exporting module parametrizations as separate and fully specialized modules.

Implementation idea:

* `Module` would receive a list of `ios` 
* There would need to be some way to check/enforce/verify the interface, i.e. ensure that module inputs are not also assigned to from within a module and that internal Signals within a module are not accessed from the outside.
* Some way to substitute a `Module` by an `Instance`.
* `build()` would export specialized modules.

whitequark, 2017-06-19:
> i.e. ensure that module inputs are not also assigned to from within a module and that internal Signals within a module are not accessed from the outside.

Right now I see the fact that I can manipulate signals without regard to module boundaries as a major feature of Migen (and I believe it's used in ARTIQ quite a bit, too). Do you suggest blanket prohibiting this, or just when exporting? In any case, splitting the Migen language into two like that seems troublesome...

jordens, 2017-06-19:
No. This shouldn't touch existing code style or features at all. It's just an trade of that one has to accept and deal with when exporting hierarchy. This will only give you the tools to do so.

mithro, 2017-10-08:
@jordens Are you actively working on this at the moment or is this something you would just like to see in the future?

jordens, 2017-10-08:
I would like to see it and don't have substantial progress.

phire, 2018-12-11:
I think this can/should be done without requiring the user to specify a list of `ios`.

Just automatically infer them from how the module is used.

whitequark, 2019-07-05:
Triage: fixed in nMigen.