m-labs/migen #73: AsyncFIFO reset

jordens, 2017-07-02:
(From #72)

Currently a reset from either the read or from the write CD resets that side's (and only that side's) address (Gray) counter.

This is probably wrong as it can lead to spurious data becoming appearing readable from the read side or/and data being lost. The latter is probably less dangerous though. Instead we probably should

1. not reset it by default and instead provide an explicit (async) reset signal to be used from either CD or both
2. reset the counters to the other side's counter (but that's a race)
3. reset both when there is a reset from either CD (similarly to what `ElasticBuffer` does)
4. choose some asymmetric behavior, i.e. have only the read side reset drop all data but the write side reset do nothing

I am leaning towards (1).

@sbourdeauducq @enjoy-digital

sbourdeauducq, 2017-07-02:
The expectation is that both domains are reset are roughly the same time, e.g. using two ``AsyncResetSynchronizer``s driven by the same signal.

whitequark, 2019-07-05:
Triage: fixed in nMigen, which natively features clock domains with asynchronous reset and synchronous release.