m-labs/migen #80: Issue with importing Signal

sheeets, 2017-10-10:
In your documentation I've seen you import in these 2 ways:
```python
>>> from migen.fhdl.std import *
or
>>> from migen.fhdl import *
```

However, when I try to do run this command:
```
>>> a = Signal(1)
NameError: name 'Signal' is not defined
```
but this works when I do:
```
>>> from migen import *
>>> a = Signal(1)
```

Is this due to some change along the way in regards to how the modules are setup?
```
>>> type(Signal())
migen.fhdl.structure.Signal
```

sbourdeauducq, 2017-10-11:
``migen.fhdl.std`` has been removed. Where in the documentation is it still mentioned?

sheeets, 2017-10-11:
It was a [tutorial](https://m-labs.hk/migen/tutorial.pdf) from 2014. Is `from migen.fhdl import *` the correct import statement to use `Signal()`?

sbourdeauducq, 2017-10-11:
No, ``from migen *``.