m-labs/migen #83: Driving a ClockDomain is broken in sim

shuffle2, 2017-10-29:
```py
        s.clock_domains.cd_spi = ClockDomain()
        s.active = Signal()
        s.bitcount = Signal(24)
        s.cmd = Signal(8)
        s.addr = Signal(24)

        s.comb += [
            s.cd_spi.clk.eq(spibus.clk),
            s.active.eq(~spibus.cs_n)
        ]

        s.sync.spi += [
            If(s.active == 0,
                s.bitcount.eq(0),
                s.cmd.eq(0),
                s.addr.eq(0)
            ).Else(
                s.bitcount.eq(s.bitcount + 1),
                If(s.bitcount < 8,
                    s.cmd.eq(Cat(spibus.si, s.cmd[:8 - 1]))
                ).Elif(s.bitcount < 8 + 24,
                    s.addr.eq(Cat(spibus.si, s.addr[:24 - 1]))
                )
            )
        ]
```
[image]

commenting the cd driver:
```py
        s.comb += [
            #s.cd_spi.clk.eq(spibus.clk)
        ]
```
[image]

You can see that `spi_clk` appears like it's always low, even though the signals depending on it are moving as if the clock is constantly running.

The SPI clk is intended to be simulated like:
```py
def spi_byte(spi, val):
    ret = 0
    for i in range(8)[::-1]:
        yield spi.si.eq(val >> i)
        yield from sys_delay(12)
        yield spi.clk.eq(1)
        yield from sys_delay(12)
        yield spi.clk.eq(0)
        ret |= (yield spi.so) << i
    yield from sys_delay(100)
    print('SPI %x -> %x' % (val, ret))
```

btw, this was only after I realized that `clocks` apparently needs to be modified into `run_simulation` (which is not documented anywhere).
```py
    run_simulation(dut, tb(dut, spi), vcd_name = 'sflash.vcd',
        clocks = {'sys': 10, 'spi': 3})
```
If `'spi'` is not passed here, all signals in `sync.spi` do not move.

jordens, 2017-10-29:
You are correct. Driving CDs in simulation does not work. You have to let the simulator drive them.

shuffle2, 2017-10-29:
So that means it's just impossible to simulate such a design with migen (anything with externally clocked logic)?

jordens, 2017-10-29:
It's not impossible but tedious. Either use a second clock driven by the simulator and some clock enable signal that you control (spi cs). Seen the urukul spi slave as an example. Or oversample (see the pdq spi slave). That better anyway since it doesn't need a CDC and can handle slow spi edges.