m-labs/migen #94: Support all function call opcodes in Python 3.5 in tracer.py

cr1901, 2017-12-13:
Per issue #62, it's an [oversight](https://github.com/m-labs/migen/issues/62#issuecomment-350777380) that `CALL_FUNCTION_KW` and `CALL_FUNCTION_VAR_KW` are not supported when searching for variable names on the stack. I think a Python 3.6 fix (#62) should depend on this being fixed first. That way common code (if any meaningful overlap) can be refactored out only once. _At present, I have not been able to construct a test case that fails._

A fix to tracer.py will need to _at least_ include supporting the `BUILD_TUPLE` opcode, which is required by [`CALL_FUNCTION_VAR_KW`](https://docs.python.org/3.5/library/dis.html#opcode-CALL_FUNCTION_VAR_KW). Current behavior would be to return `None` and stop searching for a variable name. I don't think there are any extra opcodes to support.

sbourdeauducq, 2017-12-18:
When are those opcodes emitted? Are you able to get Python to emit them at all? Emit them in a way that the tracer gets called and sees them?

cr1901, 2017-12-18:
Here is a contrived example which answers your three questions:

```
class MySpecialCSRStatus(CSRStatus):
    def __init__(self, **kwargs):
        CSRStatus.__init__(self)

class MySpecialModule(Module):
    def __init__(self, **kwargs):
        if kwargs.get("use_special", False):
            self.dummy = MySpecialCSRStatus.__init__(self, **kwargs)
        else:
            self.dummy = CSRStatus()
        

dis.dis(MySpecialModule)
print(MySpecialModule().dummy.name)
print(MySpecialModule(use_special=True).dummy.name)
```

I get the following output:

```
Disassembly of __init__:
  7           0 LOAD_FAST                1 (kwargs)
              3 LOAD_ATTR                0 (get)
              6 LOAD_CONST               1 ('use_special')
              9 LOAD_CONST               2 (False)
             12 CALL_FUNCTION            2 (2 positional, 0 keyword pair)
             15 POP_JUMP_IF_FALSE       42

  8          18 LOAD_GLOBAL              1 (MySpecialCSRStatus)
             21 LOAD_ATTR                2 (__init__)
             24 LOAD_FAST                0 (self)
             27 LOAD_FAST                1 (kwargs)
             30 CALL_FUNCTION_KW         1 (1 positional, 0 keyword pair)
             33 LOAD_FAST                0 (self)
             36 STORE_ATTR               3 (dummy)
             39 JUMP_FORWARD            12 (to 54)

 10     >>   42 LOAD_GLOBAL              4 (CSRStatus)
             45 CALL_FUNCTION            0 (0 positional, 0 keyword pair)
             48 LOAD_FAST                0 (self)
             51 STORE_ATTR               3 (dummy)
        >>   54 LOAD_CONST               0 (None)
             57 RETURN_VALUE

dummy
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-157-b7619bff0548> in <module>()
     13 dis.dis(MySpecialModule)
     14 print(MySpecialModule().dummy.name)
---> 15 print(MySpecialModule(use_special=True).dummy.name)

<ipython-input-157-b7619bff0548> in __init__(self, **kwargs)
      6     def __init__(self, **kwargs):
      7         if kwargs.get("use_special", False):
----> 8             self.dummy = MySpecialCSRStatus.__init__(self, **kwargs)
      9         else:
     10             self.dummy = CSRStatus()

<ipython-input-157-b7619bff0548> in __init__(self, **kwargs)
      1 class MySpecialCSRStatus(CSRStatus):
      2     def __init__(self, **kwargs):
----> 3         CSRStatus.__init__(self)
      4 
      5 class MySpecialModule(Module):

C:\msys64\home\william\prj\projects\misoc\misoc\interconnect\csr.py in __init__(self, size, reset, name)
    158 
    159     def __init__(self, size=1, reset=0, name=None):
--> 160         _CompoundCSR.__init__(self, size, name)
    161         self.status = Signal(self.size, reset=reset)
    162 

C:\msys64\home\william\prj\projects\misoc\misoc\interconnect\csr.py in __init__(self, size, name)
    111 class _CompoundCSR(_CSRBase, Module):
    112     def __init__(self, size, name):
--> 113         _CSRBase.__init__(self, size, name)
    114         self.simple_csrs = []
    115 

C:\msys64\home\william\prj\projects\misoc\misoc\interconnect\csr.py in __init__(self, size, name)
     35         self.name = get_obj_var_name(name)
     36         if self.name is None:
---> 37             raise ValueError("Cannot extract CSR name from code, need to specify.")
     38         self.size = size
     39 

ValueError: Cannot extract CSR name from code, need to specify.
```

You can see that the custom MiSoC object's name can be extracted thanks to `STORE_ATTR`, but it isn't b/c of the function opcode used:

```
30 CALL_FUNCTION_KW         1 (1 positional, 0 keyword pair)
33 LOAD_FAST                0 (self)
36 STORE_ATTR               3 (dummy)
```

I don't think anything in Migen/MiSoC itself will trigger this behavior. I haven't seen any Migen class that takes `**kwargs` and/or is called from a function that supplies said class with the `**kwargs` dictionary, which is how `CALL_FUNCTION{_VAR}_KW` would be emitted. So whether you support code like this at all is up to you. But it _can_ be done.

sbourdeauducq, 2017-12-19:
Supporting this sounds logical.

cr1901, 2017-12-20:
Not sure why the issue wasn't closed automatically, despite me putting "Closes #94" in PR #97...

closed 2017-12-20 by cr1901