m-labs/migen #121: Not able to simulate

bhatmahadev6, 2018-07-29:
Traceback (most recent call last):
  File "ORg.py", line 11, in <module>
    dut = ORGate()
  File "ORg.py", line 3, in __init__
    self.a = Signal()
NameError: name 'Signal' is not defined
//for the example given in the userguide

sbourdeauducq, 2018-07-30:
Did you import migen?

bhatmahadev6, 2018-07-30:
Please give your whatsapp no.or email id I wanted to work with Migen for my
small project
I tried importing it says no module named migen.fhdl.std

sbourdeauducq, 2018-07-30:
You need to get the import to work first. How did you install migen? Are there any stale copies in your system?

bhatmahadev6, 2018-07-30:
i installed migen using python3 setup.py command

bhatmahadev6, 2018-07-30:
how can i remove stale copies

bhatmahadev6, 2018-07-30:
its saying no module named migen.fhdl.std

rroohhh, 2018-07-30:
you need to use ```from migen import *```, the tutorial you used is outdated.

closed 2018-07-31 by bhatmahadev6

PriyaJ-coder, 2021-11-17:
> you need to use `from migen import *`, the tutorial you used is outdated.

Hey, I am also getting the same error 'module not found'....... can you pls tell me how did you solve the problem?

bhatmahadev6, 2021-11-18:
try with different Versions of Python maybe 2 or 3.

sbourdeauducq, 2021-11-18:
Only Python 3 is supported.