m-labs/migen #122: How to link between Migen and Icarus Iverilog For simulation??

bhatmahadev6, 2018-07-31:
As given in docs i am not able to understand how to link to iverilog and simulate it further in gtkwave.
I am not able to understand "make" commands in docs

sbourdeauducq, 2018-07-31:
The docs you are reading are out of date, there is the migen built-in simulator now that does not require an external simulator. If you do want to use Icarus, you can resurrect the old migen code.

closed 2018-07-31 by sbourdeauducq

bhatmahadev6, 2018-07-31:
from migen import *
from migen.sim.vcd import VCDWriter, DummyVCDWriter
class ORGate(Module):
  def __init__(self):
    self.a = Signal()
    self.b = Signal()
    self.x = Signal()

    ###

    self.comb += self.x.eq(self.a | self.b)

dut = ORGate()

def testbench():
  yield dut.a.eq(0)
  yield dut.b.eq(0)
  yield
  assert (yield dut.x) == 0

  yield dut.a.eq(0)
  yield dut.b.eq(1)
  yield
  assert (yield dut.x) == 1

run_simulation(dut, testbench(),vcd_name="file.vcd")

i Ran this code ,But i am not getting any output,its just executing

bhatmahadev6, 2018-07-31:
sorry Got it...