m-labs/migen #136: Can't set signal with subsingals equal to a pad

tdaede, 2018-09-27:
If I define some signals in a module:
```       
        self.p1 = Signal()
        self.p1.s = Signal()
```
and then assign it to a platform pad with subsignals, it works:
```
m.comb += plat.request("p1").eq(m.p1)
```
But if I do the opposite, e.g. I want p1 to be an input:
```
m.comb += m.data_pad.eq(plat.request("data"))
```
then I get an error:
```
TypeError: Object '<Record u:d:l:r:s:b:c at 0x7ffb1d9c3d30>' of type <class 'migen.genlib.record.Record'> is not a Migen value
```

cr1901, 2018-09-27:
IMO, you really shouldn't be assigning your own attributes to a `Signal` like the following lines:

```
self.p1 = Signal()
self.p1.s = Signal()
```

If you want structure, you should use a `Record`.

The problem here is that the return value of `plat.request()` for something with `Subsignals` is a `Record` type. `Record` comes with its own `eq()` [overload](https://github.com/m-labs/migen/blob/master/migen/genlib/record.py#L111) that understands `Record` types. _It just so happens to work for `Signals` with custom attributes anyway that aren't of type `Record`._ This is what's happening here:

```
m.comb += plat.request("p1").eq(m.p1)
```

On the other hand, on the next line:

```
m.comb += m.data_pad.eq(plat.request("data"))
```

`data_pad` is still a `Signal`, and so it calls the `Signal.eq()` [method](https://github.com/m-labs/migen/blob/master/migen/fhdl/structure.py#L153). Ultimately, `Signal.eq()` calls `wrap` [internally](https://github.com/m-labs/migen/blob/master/migen/fhdl/structure.py#L153) and that doesn't support `Record` types :). Furthermore, `Signal.eq()` wouldn't know anything about your custom attribute `s` either :).

The solution is to make `data_pad` a `Record`, _and_ to make `self.p1` a `Record`. I don't know the specifics of your use case, so see this [example file](https://github.com/m-labs/migen/blob/master/examples/basic/record.py) for usage.

_That said_, we should probably check that the type passed to `Record.eq()` is in fact another `Record` type and not a `Signal` masquerading as a `Record` type.

whitequark, 2019-07-05:
Triage: in nMigen, `Record`s are ordinary values like `Signal`s and can be assigned. As for custom attributes, it is unfortunate that they are silently ignored here, but disallowing them with something like `__slots__` seems like it would be a serious problem if someone wanted to store auxiliary data attached to a `Signal`, so it should probably not be done.

closed 2019-07-05 by whitequark