m-labs/migen #137: Python logical operator 'and' silently dropped

tdaede, 2018-09-27:
The following migen snippet:
```
               If((self.addr == 0x18a) and (self.joy_enabled == 1),
```
produces the following Verilog:
```
                if ((addr_1 == 9'd394)) begin
```

sbourdeauducq, 2018-09-27:
Yes, you cannot use ``and`` - it cannot be redefined in Python.

sbourdeauducq, 2018-09-27:
Actually it should raise ``TypeError("Attempted to convert Migen value to boolean")`` (see fhdl/structure.py). Can you find out why that does not happen?

whitequark, 2018-11-10:
@sbourdeauducq Because of the special case in `__bool__`:
```
        # Special case: Constants and Signals are part of a set or used as
        # dictionary keys, and Python needs to check for equality.
        if isinstance(self, _Operator) and self.op == "==":
            a, b = self.operands
            if isinstance(a, Constant) and isinstance(b, Constant):
                return a.value == b.value
            if isinstance(a, Signal) and isinstance(b, Signal):
                return a is b
            if (isinstance(a, Constant) and isinstance(b, Signal)
                    or isinstance(a, Signal) and isinstance(b, Constant)):
                return False
```
I don't understand why it lives in `__bool__` though.

sbourdeauducq, 2018-11-12:
Where should it be instead?

whitequark, 2018-11-12:
@sbourdeauducq Is it realistic to make only `Signal` and `Constant` hashable in the first place? AFAICT Migen does not rely on other values being hashable, but I might be wrong.

sbourdeauducq, 2018-11-12:
Maybe, can you test it (including with ARTIQ)?

whitequark, 2018-11-12:
OK

xobs, 2019-04-18:
I also just ran into this problem:

```python
class OrderTest(Module):
    def __init__(self):
        counter = Signal(8)
        sig = Signal()
        self.sync += [
            counter.eq(counter + 1),
            If((counter == 1) or (counter == 2) or (counter == 3),
                sig.eq(1)
            )
        ]

print(str(verilog.convert(OrderTest())))
```

This results in:

```verilog
...
      if ((counter == 2'd3)) begin
                sig <= 1'd1;
      end
...
```

whitequark, 2019-04-18:
FYI this (and many other issues) are fixed in https://github.com/m-labs/nmigen/. But nMigen is not yet quite ready.

whitequark, 2019-07-05:
Triage: fixed in nMigen.

DurandA, 2019-10-29:
@sbourdeauducq @whitequark What is the correct way of doing logical AND or logical OR in Migen then? `and_(a, b)`? I couldn't find any example with something similar to `_Operator("and", [a, b])` or `_Operator("||", [a, b])` in the repository.

jordens, 2019-10-29:
`(x == 2) & (y > 9)`

DurandA, 2019-10-29:
> `(x == 2) & (y > 9)`

This is the [bitwise operator](https://personal.utdallas.edu/~jxw143630/index_files/Page5212.htm), which [does not behave like the logical operator](https://www.quora.com/In-Verilog-what-is-the-difference-between-logical-and-bit-wise-operator).

whitequark, 2019-10-30:
In nMigen you can do `x.bool() & y.bool()` to get an equivalent of the logical AND that works regardless of the width of `x` and `y`.

DurandA, 2019-10-30:
Thanks for your answers! Is there a bool() equivalent in Migen or should we use something like `reduce(or_, a)`?

jordens, 2019-10-30:
`a != 0`