m-labs/migen #141: SyncFIFO does not infer SRL16/32

JohnSully, 2018-10-09:
This is in reference to the LiteDRAM SyncFIFOs in each bank machine as seen here: https://github.com/enjoy-digital/litedram/blob/master/litedram/core/bankmachine.py

For a basic dynamic shift register the most efficient packing in a Xilinx part is an SRL16 or 32 which can fit in a single slice.  It seems that the toolchain is not able to infer these based on the SyncFIFO logic.  This leads to inefficient usage for large FIFOs.  A lot is being left on the table.

The code as used in LiteDRAM doesn't use any fancy features that would prevent use of a SRL16, but registers are being inferred instead.

JohnSully, 2018-10-09:
I'm researching into why.

An SRL32 is a Serial->Parallel shift register, so you can index it to turn it into a FIFO.  The important thing is that data is fed in serially and you read out based upon the depth counter, doing the reverse prevents inferring one.  Also the data can't be reset (but of course the depth counter can).

sbourdeauducq, 2018-10-09:
Isn't it the "distributed RAM" mode that you are describing? That should be OK for SyncFIFO.

JohnSully, 2018-10-09:
More info on page 34 of UG474: https://www.xilinx.com/support/documentation/user_guides/ug474_7Series_CLB.pdf

Distributed memory is one mode, and SRL is another.  Right now in the case of LiteDRAM its being inferred as a series of registers, with a buffer depth of 8 that's 4 slices, where it would only use half if it were an SRL16.

Ensuring these are inferred shrinks designs dramatically in my experience.

JohnSully, 2018-10-09:
I was incorrect in my previous message, I was seeing buffer registers.  Its constructed as a ring buffer with SLICEMs.  This design is a good fit for Altera devices or large buffers on Xilinx.

With an SRL style design you would always insert at 0, and so you'd only need a depth register.  I'll modify it and test the results on LiteDRAM to see how big a difference it makes.

sbourdeauducq, 2018-10-09:
Note that the "depth register" design cannot be mapped to block RAM and is not appropriate for deep FIFOs.

JohnSully, 2018-10-09:
Agreed.  Its really only appropriate for FIFOs of depth 32 or less on Xilinx parts.  Still it might be appropriate for conditional compilation in the right conditions.  I'm coding up a version in migen and will test its results with LiteDRAM which should be a nice real-world example to see its effect.

The main issue I'm trying to solve is inefficient packing in LiteDRAM.

JohnSully, 2018-10-09:
It seems its not possible to have migen generate verilog that can ever infer an SRL.  An array generates:
```
reg [23:0] sdram_bankmachine6_cmd_buffer_lookahead_rgbuffer0 = 24'd0;
reg [23:0] sdram_bankmachine6_cmd_buffer_lookahead_rgbuffer1 = 24'd0;
...
```
And the tools aren't able to infer from shift registers made from independent signals.  

It needs to be in the form:
reg [FIFO_DEPTH-1:0][DATA_WIDTH-1:0] rgbuffer;
and either a for loop or rgbuffer <= { rgbuffer[FIFO_DEPTH-2:0], in_data }; for the actual shift operation.

sbourdeauducq, 2018-10-09:
What about having many 1-bit signals?

JohnSully, 2018-10-09:
I think that would work, but a further issue is the data in an SRL can't have a reset.  reset=None fails to compile.  I'm trying to see if there is a way to prevent a reset from getting generated.

JohnSully, 2018-10-09:
Synthesis Numbers:

`
  | Without | With
-- | -- | --
LUT | 3051 | 3020
LUTRAM | 411 | 459
FF | 2668 | 2596
`
But more importantly on implementation it saved me 33 slices which is 5%.  My guess is the main benefit is a reduction in control sets.

jordens, 2018-10-09:
IIRC one general issue I remember with inferring Xilinx SRLs from Migen is that they need to be `reset_less`. May or may not be relevant here though.

jordens, 2018-10-09:
Oh. I see you had already noticed. Sorry for the noise.

whitequark, 2019-07-05:
Triage: it looks like the main issue here was that Xilinx SRLs are only inferred from `reset_less` signals, and it works with that, so I see this as solved. Let me know if there's something else that needs to be done.

closed 2019-07-05 by whitequark