m-labs/migen #142: Negative initialization produces incorrect verilog

JohnSully, 2018-10-09:
Migen: 
self.idx = Signal(max=depth, reset=-1)

Generated Verilog:
reg [3:0] sdram_bankmachine0_cmd_buffer_lookahead_idx = 4'd-1;

The assigned value should be -4'd1 instead to be valid Verilog.  Not my favourite thing about the language.

sbourdeauducq, 2018-10-09:
Your signal should be signed for a negative value to be legitimate. Migen should print an error here instead of emitting invalid verilog, maybe at the minimum add an assert in ``_printconstant``.

sbourdeauducq, 2018-10-09:
Though this isn't consistent with what happens when assigning a negative value to an unsigned signal; so maybe ``_printconstant`` should handle it.

JohnSully, 2018-10-09:
It's probably horrible practice, but I do it to set a signal to 0xFF regardless of its length.  The rest of the code uses it unsigned.

JohnSully, 2018-10-09:
Signal(max=depth, ~0) produces the same incorrect code.  What's the right way to initialize a signal to all 1s?

sbourdeauducq, 2018-10-09:
2**depth-1

JohnSully, 2018-10-09:
I think it would have to be 2**(ceil(log2(depth))-1 right?

-1 or ~0 is much easier

sbourdeauducq, 2018-10-09:
OK.
~0 == -1 in Python.

whitequark, 2019-07-05:
Triage: fixed in nMigen. In particular, `~0` and `-1` initialization values for signed as well as unsigned signals produce the expected result.