m-labs/migen #153: Create experimental migen using edalize as part of migen.build

mithro, 2018-10-28:
Let's use this issue to talk / think further about the idea of using edalize as part of migen.

----

@sbourdeauducq [mention in #149](https://github.com/m-labs/migen/pull/149#issuecomment-433579421)

> IMO the migen build system needs an overhaul. It is something that was initially written to manage simple designs on Xilinx ISE, and then grew out in a somewhat anarchistic manner. Collecting specifications from different people on what it should and should not do, and then deciding on a nice architecture and implementing it, sounds like a good way forward. I would also try to decouple it from `Record` (i.e. not implement features in Record only to overcome limitations of the build system).

To which [I (@mithro) replied](https://github.com/m-labs/migen/pull/149#issuecomment-433582337);
> @sbourdeauducq I have something to sell you!
> 
> At OrConf 2018 it was clear that there are a _lot_ of Python tools. Pretty much all these tools need to interface with various EDA tools. It quickly became clear that it would be really useful to have a _good_ library for interfacing with EDA tools. I started trying to collect a whole bunch of random notes at http://bit.ly/pyfgpalibs
> 
> @sbourdeauducq it would be good to get your feedback on the info there, specifically around how you think the current build system needs to change.
> 
> @olofk who created FuseSoC, has taken a lead in trying to create this "good" Python library. It current goal is to [extract the library out of FuseSoC and is calling it EDAlize](https://github.com/olofk/fusesoc/tree/master/edalize). Would you have time to provide him with feedback on how that is going and how it would make sense to integrate with Migen? I think if EDAlize was used by Migen it would go a long way to making sure the library is actually _really_ useful and give confidence for others to also use it.
> 
> FYI I'm personally excited about the idea of having a common library that I can then do things like;
> 
> * Add support for extracting things like messages like errors and warning.
> * Reporting of resource usages (in a common programmatic manner).
> * Better way to provide "console gui" for the progress of the build (that is compatible with Travis).
> * Better way to provide "cut back" versions of toolchain. (See stuff at https://github.com/timvideos/litex-buildenv/tree/master/.travis).

And [@olofk also said](https://github.com/m-labs/migen/pull/149#issuecomment-433655389);
> Migen is definitely one of the projects I have envisioned using edalize, so I'm very interested in feedback here. To begin with, I have now moved it to a [separate repository](https://github.com/olofk/edalize). I have also relicensed it to BSD. This was the plan all along and I have collected acks for relicensing from all contributors. And no, it doesn't handle pin allocation, so I'm aware that this might further sidetrack the original bug report, but it still felt like the proper place to reply

mithro, 2018-10-28:
It would also be good to compare the [litex.build](https://github.com/enjoy-digital/litex/tree/master/litex/build) features. 

[@enjoy-digital said the following recently](https://github.com/enjoy-digital/litex/issues/110#issuecomment-426899584);
> There are some specific features in `litex.build` and some of them are probably not going to land in `migen.build`. I 'll try to merge the new features that could be useful in `migen.build`, but i want to keep `litex.build` for at least my projects. (otherwise i would have to maintain a fork of migen which is not better). Note that you can use only use `migen.build` in `litex-buildenv.` if your platform is described using `migen.build` elements.

mithro, 2018-10-28:
I have a [WIP pull request](https://github.com/enjoy-digital/litex/pull/114) for getting `litex.build` to allow using Yosys + Vivado.

The aim was that as we are starting to get close to having [SymbiFlow](https://symbiflow.github.io/) be potentially usable for (very simple) LiteX based SoCs, it would be good to start finding any potential Yosys bugs independently from SymbiFlow bugs.

olofk, 2018-10-29:
FYI, I implemented support for synthesis with yosys in the Edalize vivado backend now https://github.com/olofk/edalize/tree/yosys-vivado

whitequark, 2018-12-14:
I have decided against using Edalize in Migen/nMigen because it relies on (GNU?) Make being available. This results in generally poor user experience on Windows.

mithro, 2018-12-14:
Maybe @olofk would accept a patch to make it possible to use Edalize without make?

olofk, 2018-12-14:
This is a fair criticisms and I have been thinking about ways to improve this. Using make isn't really the goal, but just having a way to easily (re)build the project. I use this a lot myself when delivering to clients so that they don't need FuseSoC/edalize on their systems.

As for alternatives, I'm happy with moving to sh/bat, but also would like to know if anyone has opinions about using ninja instead.

Also, for some backends, at least verilator, we need make anyway, but that could stay as a tool-specific dependency

whitequark, 2018-12-14:
> As for alternatives, I'm happy with moving to sh/bat, but also would like to know if anyone has opinions about using ninja instead.

Ninja has essentially the same issues as make: it is an additional thing to install not only on Windows but *also* on Linux. Plus neither make nor ninja provide a lot of benefit for Edalize AFAIK, most HDL flows end up always rebuilding everything.

whitequark, 2019-07-05:
This was investigated as a part of nMigen platform (re)design, and the conclusion was that Edalize does not offer much benefit in this area.

closed 2019-07-05 by whitequark