m-labs/migen #159: SyncFIFO replace signal not ASIC compatible

Fatsie, 2018-11-11:
The replace signal of SyncFIFO is not properly initialized during a reset making it incompatible with ASIC implementation. Given following code:
```python
from migen.genlib.fifo import SyncFIFO
from migen.fhdl.verilog import convert

fifo = SyncFIFO(8, 4)
convert(fifo, name="fifo").write("fifo.v") 
```
This generates `fifo.v` with line:
```verilog
reg replace = 1'd0;
```
But `replace` is not given new value when sys_rst is high. This makes the generated verilog incompatible with ASIC implementation. This problem was found by doing simulation on output generated with `asic_syntax=True`.

sbourdeauducq, 2018-11-11:
``asic_syntax`` will be removed. Is there any problem without? Why do you need it?

Fatsie, 2018-11-11:
As said in the issue the problem is not `asic_syntax`; it's that the generated verilog code is not ASIC compatible. Using `asic_syntax` will also show the problem in simulation and not only when the chip comes back from the fab...
The problem is that `replace` is not reset to 0 during `sys_rst`.

sbourdeauducq, 2018-11-11:
``replace`` is an input to the FIFO. If you are not using it, then Migen ties it to 0, just like it does with the other control and data input signals of the FIFO, e.g. ``we`` and ``re``.

Fatsie, 2018-11-11:
Quick check with yosys seems to confirm that it does the right thing. Will do extensive post-synthesis simulations later on.

closed 2018-11-11 by Fatsie