m-labs/migen #171: please add install instruction to readme?

TG-Techie, 2018-12-18:
The title almost says it all,  I had to look this up to figure out how to install it: http://jeffrey.co.in/blog/2014/01/installing-migen/

the installation was easy but finding out how was much less so.

sbourdeauducq, 2018-12-19:
Well OK, but e.g. VPI has been removed. For MacOS you can use nix with this:
https://github.com/m-labs/artiq/blob/master/nix/pkgs/python3Packages.nix#L51
or the Anaconda package.

jordens, 2018-12-19:
For those that don't have experience installing and using Python packages, `pip install git+https://github.com/m-labs/migen.git` is probably all they want to know.
The others will be just fine.

sam-falvo, 2019-01-06:
I've had great luck so far with this approach as well:

$ virtualenv3 env3
$ . env3/bin/activate
$ pushd $PATH_TO_MIGEN
$ ./setup.py install
$ popd

TG-Techie, 2019-12-15:
it now works great with pip3 !

closed 2019-12-15 by TG-Techie