m-labs/migen #175: `collections` is deprecated in python 3.7

xobs, 2019-02-18:
If you use migen under Python 3.7+, you get this confusing error message:

```
D:\Code\fomu\foboot-bitstream\deps\migen\migen\fhdl\structure.py:484: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
```

It took me a while to discover that `collections` is apparently a module from Python, and the `3.8` they refer to is Python 3.8 (and not `migen 3.8`, which is what the error implies).

xobs, 2019-02-18:
This is a patch that appears to fix the issue, but I'm sure there's a better way to do it:

```patch
From 0add236553e0f12982dd70522442bae6cffc3c32 Mon Sep 17 00:00:00 2001
From: Sean Cross <sean@xobs.io>
Date: Mon, 18 Feb 2019 15:42:28 +0800
Subject: [PATCH] collections: hack to fix python 3.7

When run under Python 3.7, the interpreter emits warnings about
`collections` being deprecated under 3.8.

This patch hacks migen to use collections or collections.abc as necessary.

---
 migen/fhdl/module.py    | 2 +-
 migen/fhdl/structure.py | 5 +++--
 migen/sim/core.py       | 5 +++--
 migen/util/misc.py      | 2 +-
 4 files changed, 8 insertions(+), 6 deletions(-)

diff --git a/migen/fhdl/module.py b/migen/fhdl/module.py
index 7a20993..135d2ba 100644
--- a/migen/fhdl/module.py
+++ b/migen/fhdl/module.py
@@ -1,4 +1,4 @@
-import collections
+import collections.abc as collections
 from itertools import combinations
 
 from migen.util.misc import flat_iteration
diff --git a/migen/fhdl/structure.py b/migen/fhdl/structure.py
index bb0a258..0865e76 100644
--- a/migen/fhdl/structure.py
+++ b/migen/fhdl/structure.py
@@ -1,4 +1,5 @@
 import builtins as _builtins
+import collections.abc as _collections_abc
 import collections as _collections
 import re as _re
 
@@ -481,7 +482,7 @@ class _Assign(_Statement):
 
 
 def _check_statement(s):
-    if isinstance(s, _collections.Iterable):
+    if isinstance(s, _collections_abc.Iterable):
         return all(_check_statement(ss) for ss in s)
     else:
         return isinstance(s, _Statement)
@@ -588,7 +589,7 @@ class Case(_Statement):
             if (not isinstance(k, Constant)
                     and not (isinstance(k, str) and k == "default")):
                 raise TypeError("Case object is not a Migen constant")
-            if not isinstance(v, _collections.Iterable):
+            if not isinstance(v, _collections_abc.Iterable):
                 v = [v]
             if not _check_statement(v):
                 raise TypeError("Not all objects for case {} "
diff --git a/migen/sim/core.py b/migen/sim/core.py
index bf2b95f..4a5e78c 100644
--- a/migen/sim/core.py
+++ b/migen/sim/core.py
@@ -1,4 +1,5 @@
 import operator
+import collections.abc as collections_abc
 import collections
 import inspect
 from functools import wraps
@@ -227,7 +228,7 @@ class Evaluator:
                         break
                 if not found and "default" in s.cases:
                     self.execute(s.cases["default"])
-            elif isinstance(s, collections.Iterable):
+            elif isinstance(s, collections_abc.Iterable):
                 self.execute(s)
             elif isinstance(s, Display):
                 args = []
@@ -279,7 +280,7 @@ class Simulator:
         self.generators = dict()
         self.passive_generators = set()
         for k, v in generators.items():
-            if (isinstance(v, collections.Iterable)
+            if (isinstance(v, collections_abc.Iterable)
                     and not inspect.isgenerator(v)):
                 self.generators[k] = list(v)
             else:
diff --git a/migen/util/misc.py b/migen/util/misc.py
index 0eadc2b..8222c9f 100644
--- a/migen/util/misc.py
+++ b/migen/util/misc.py
@@ -1,5 +1,5 @@
 from math import gcd
-import collections
+import collections.abc as collections
 
 
 def flat_iteration(l):
-- 
2.7.4

```

sbourdeauducq, 2019-02-18:
@xobs This patch is quite hacky. Can you clean it up by:
* *adding* ``import collections.abc``
* removing ``import collections`` when necessary
* using ``collections.abc`` where relevant, e.g. ``if (isinstance(v, collections.Iterable)`` → ``if (isinstance(v, collections.abc.Iterable)``

xobs, 2019-02-19:
@sbourdeauducq Sure, I just wasn't sure if this was the preferred approach, and so I just came up with something that worked.

I'll make a proper patch and create a PR.

closed 2019-02-21 by sbourdeauducq