m-labs/migen #179: Conda installation fails

chipmuenk, 2019-04-28:
When trying to install migen under conda (Win7 64bit, latest Anaconda, Python 3.7) using

`conda install -c m-labs migen=0.8`

I get the strange error

```
UnsatisfiableError: The following specifications were found to be in conflict:
  - conda[version='>=4.6.14']
  - migen=0.8
```
Has someone else had this problem or could this be a problem with my Anaconda install?

sbourdeauducq, 2019-04-29:
Try rebuilding the conda package from https://github.com/m-labs/migen/tree/master/conda.

chipmuenk, 2019-04-30:
You mean cloning migen to my computer and building it locally? I could try that but the problem is that in the not-so-long run I would like to integrate migen with my filter design tool pyFDA (https://github.com/chipmuenk/pyfda) for simulating and synthesizing fixpoint filters. For a start, I could distribute pyFDA via PyPI and Conda without migen and only enable migen features when the module is found - that's the way I do it with myHDL at the moment. But having migen in the requirements.txt file would be much easier and nicer :-)

sbourdeauducq, 2019-05-01:
pyFDA looks cool :)
I meant cloning the Migen repository, and building the Migen *conda package* locally, which you can then upload to your channel on anaconda.org (or your own site) alongside pyFDA.
I suspect that the error is due to our package being built with a different conda version than yours, and building your own package should solve this.
Just a few words of warning, we've been using conda intensively for a number of years and it buggy and a real PITA. Unfortunately, it seems you are using Windows and Qt so there isn't really a better solution at the moment.
Good luck and just let me know if you need changes to the upstream migen conda recipe - we're not using this conda package anymore, so we can make whatever changes you want there.

chipmuenk, 2019-05-01:
Thanx :-)
Yes, I also meant building conda package locally, sorry for the confusion. I do agree with the PITA, especially when it comes to conda develop.
I'm using both Windows and Linux, so I need cross-platform solutions. I'm using pyFDA for my DSP courses (and the students have all sorts OS) so I need to make sure that deployment and installation is as smooth as possible. What is the recommended way to install migen - simply cloning the repo?

A few days ago I merged a PR for a recipe to create a pyFDA Flatpak which might be a good solution for Linux users. A pip package of migen would be really helpful for that purpose; am I right that there is none at the moment? Would you mind if I tried to create a "migen-core" pip package on PyPI without the FPGA board constraint files etc.? The version number would be identical to the cloned migen version. 

I'm looking forward to explore the options of migen, to me it seems more user friendly than myhdl and more suited to my needs (fully synchronous DSP systems, targeted for FPGAs).

sbourdeauducq, 2019-05-01:
> What is the recommended way to install migen - simply cloning the repo?

Cloning the repos and setting PYTHONPATH does work. You can also run ``setup.py`` from the repos manually. I do not know of an excellent method for installing Migen on Windows; maybe conda, pip (it can install from Git using ``pip install git+https://github.com/m-labs/migen.git``), or manual installation with setup.py are the least bad ones.

> Would you mind if I tried to create a "migen-core" pip package on PyPI without the FPGA board constraint files etc.?

Not at all, please go ahead.
FYI, I've been using Nix recently, which works really well despite the learning curve and somewhat lacking documentation. See https://git.m-labs.hk/M-Labs/nix-scripts/src/branch/master/artiq/pkgs/python-deps.nix#L91

chipmuenk, 2019-05-02:
BTW, is there any specific reason for not providing a migen package on PyPI (except of course for "well, somebody's got to do it" :-) )?

sbourdeauducq, 2019-05-02:
No other reason. Providing it includes pushing new releases, so, it's quite a bit of boring work.

sbourdeauducq, 2019-07-05:
Any action to take on this issue (e.g. someone wants to maintain a conda package), or can we close it?

chipmuenk, 2019-07-05:
Sorry, forgot about that one!

closed 2019-07-05 by chipmuenk

timkpaine, 2023-03-18:
https://github.com/conda-forge/staged-recipes/pull/22321