m-labs/migen #192: IOBuf inference only generate first bit of vector

TomKeddie, 2019-10-25:
Using this code
```
class GPIOBidirectional(Module, AutoCSR):
    def __init__(self, soc, pad_name):
        pins_pad = soc.platform.request(pad_name)
        pins_t = TSTriple(len(pins_pad))

        self.specials += pins_t.get_tristate(Cat(pins_pad.p0, pins_pad.p1, pins_pad.p2, pins_pad.p3, pins_pad.p4, pins_pad.p5, pins_pad.p6, pins_pad.p7))
        self._pins_in = CSRStatus(len(pins_pad))
        self._pins_out = CSRStorage(len(pins_pad))
        self._pins_oe = CSRStorage(len(pins_pad))

        self.specials += MultiReg(pins_t.i, self._pins_in.status)
        self.comb += pins_t.o.eq(self._pins_out.storage)
        self.comb += pins_t.oe.eq(self._pins_oe.storage)

```
(which I plan to make generic and add to litex)
the generated verilog only has a single bit for the tristate control.
```
wire [7:0] pmodc_pins_oe_storage;
wire [7:0] pmodc_o;
wire pmodc_oe;
wire [7:0] pmodc_i;
wire [7:0] pmodc_status;
.
.
assign pmodc_oe = pmodc_pins_oe_storage;
.
.
assign {pmodc_p7, pmodc_p6, pmodc_p5, pmodc_p4, pmodc_p3, pmodc_p2, pmodc_p1, pmodc_p0} = pmodc_oe ? pmodc_o : 8'bz;
assign pmodc_i = {pmodc_p7, pmodc_p6, pmodc_p5, pmodc_p4, pmodc_p3, pmodc_p2, pmodc_p1, pmodc_p0};
```
then vivado warns, and an IOBUF is only generated for the first bit.
```
WARNING: [DRC RPBF-3] IO port buffering is incomplete: Device port pmodc_p1 expects both input and output buffering but the buffers are incomplete.
WARNING: [DRC RPBF-3] IO port buffering is incomplete: Device port pmodc_p2 expects both input and output buffering but the buffers are incomplete.
WARNING: [DRC RPBF-3] IO port buffering is incomplete: Device port pmodc_p3 expects both input and output buffering but the buffers are incomplete.
WARNING: [DRC RPBF-3] IO port buffering is incomplete: Device port pmodc_p4 expects both input and output buffering but the buffers are incomplete.
WARNING: [DRC RPBF-3] IO port buffering is incomplete: Device port pmodc_p5 expects both input and output buffering but the buffers are incomplete.
WARNING: [DRC RPBF-3] IO port buffering is incomplete: Device port pmodc_p6 expects both input and output buffering but the buffers are incomplete.
WARNING: [DRC RPBF-3] IO port buffering is incomplete: Device port pmodc_p7 expects both input and output buffering but the buffers are incomplete.
```
If I hack the verilog

This works as expected
```
assign pmoda_p7 = pmoda_oe[7] ? pmoda_o[7] : 1'bz;
assign pmoda_p6 = pmoda_oe[6] ? pmoda_o[6] : 1'bz;
assign pmoda_p5 = pmoda_oe[5] ? pmoda_o[5] : 1'bz;
assign pmoda_p4 = pmoda_oe[4] ? pmoda_o[4] : 1'bz;
```
This does not (vivado bug?)
```
wire [7:0] pmodc_oe;
.
.
assign {pmodc_p7, pmodc_p6, pmodc_p5, pmodc_p4, pmodc_p3, pmodc_p2, pmodc_p1, pmodc_p0} = pmodc_oe ? pmodc_o : 8'bz;
```

TomKeddie, 2019-10-25:
Full source can be found at https://github.com/TomKeddie/prj-litex/blob/master/arty/gpio/hw/arty-gpio.py

TomKeddie, 2019-11-10:
This looks to be caused by the way the oe signal is contructed, this is probaly intentional, I guess a param could be added to init to override this behaviour?
```
class TSTriple:
    def __init__(self, bits_sign=None, min=None, max=None, reset_o=0, reset_oe=0, reset_i=0,
                 name=None):
        self.o = Signal(bits_sign, min=min, max=max, reset=reset_o,
                        name=None if name is None else name + "_o")
        self.oe = Signal(reset=reset_oe,
                         name=None if name is None else name + "_oe")
        self.i = Signal(bits_sign, min=min, max=max, reset=reset_i,
                        name=None if name is None else name + "_i")
```

whitequark, 2019-11-10:
That's how TSTriple is designed; in your use case, it would be appropriate to use one TSTriple per GPIO bit.

closed 2019-11-10 by TomKeddie