m-labs/migen #207: Problem with copy_sources(self, build_dir, subdir="imports")

andresurian, 2020-04-05:
I work with a xilinx FPGA (spartan 6), with the ise toolchain, when i try to run the lab002 from fpga_101 i have an xst problem: ERROR: Xst: 2927 - "/home/hyde/Digital/Soc/fpga_101/lab002/build/top.prj "line 1: Source file imports / home / hyde / Digital / Soc / fpga_101 / lab002 / bcd.v does not exist

This is because the file address is not taken well, so modify the generic_platform.py file on line 325: def copy_sources (self, build_dir, subdir = os.getcwd () + "/build/build/imports"):

In this way i solve the error
[generic_platform.txt](https://github.com/m-labs/migen/files/4435236/generic_platform.txt)

sbourdeauducq, 2020-04-06:
I do not understand the problem. Can you post the source code of a minimal Migen design that reproduces the issue? We are using this ``copy_sources`` function all the time and have not seen problems.

Alain94W, 2020-11-17:
I had the same issue, the copy_sources function try to copy an external v file into a non existing directory : imports/homefolder/project/...
The external v file is imported using : platforme.add_source("spislave.v","verilog","work") 

eg, prj file content : 
`verilog work pifpdl2.v
verilog work imports/home/xx/Dev/spislave.v`

I also solve it using a slightly different manner (with the debug print) : 
`def copy_sources(self, build_dir, subdir="imports"):#"imports"
        copied_sources = set()

        for filename, language, library in self.sources:
            path = _make_local_path(subdir, filename)

            # source filenames are assumed relative to the build_dir
            print("Build dir:",os.path.dirname(os.path.abspath(build_dir)))
            src = os.path.join(build_dir, filename)
            print("Source :",src)
            print("Path :",path)
            # copy to path that starts with build_dir
            dest = os.path.join(os.path.dirname(os.path.abspath(build_dir)), path)
            print("Dest : ",dest)
            os.makedirs(os.path.dirname(dest), exist_ok=True)
            shutil.copyfile(src, dest)

            # return entries relative to build_dir
            copied_sources.add((path, language, library))

        return copied_sources`

After doing this change : 

`verilog work pifpdl2.v
verilog work imports/spislave.v`

Another point now is that randomly in the prj file, the imported v file appear to be the first, then the compiler is not working.
You can make a test by trying to import a v file in your design with spartan 6 and ise toolchain.

eg sometime it write this in the prj file :
 
`verilog work imports/spislave.v
verilog work pifpdl2.v
`
And this is not compiling.

sbourdeauducq, 2020-11-18:
@alain94w please post minimal project that exhibits the problem.

Alain94W, 2020-11-19:
I will do that as soon as possible, as a new father, it's difficult to find time for hobby ...

smunaut, 2021-06-28:
Got the same issue ...

The root problem is that by the time `copy_sources` is called, the platform has already did a `os.chdir(build_dir)`. And so re-adding the `build_dir` to the copy destination screws things up.

smunaut, 2021-06-28:
Fixed it with :

```diff
diff --git a/migen/build/generic_platform.py b/migen/build/generic_platform.py
index c07883b..d09acbe 100644
--- a/migen/build/generic_platform.py
+++ b/migen/build/generic_platform.py
@@ -331,7 +331,7 @@ class GenericPlatform:
             # source filenames are assumed relative to the build_dir
             src = os.path.join(build_dir, filename)
             # copy to path that starts with build_dir
-            dest = os.path.join(build_dir, path)
+            dest = path
             os.makedirs(os.path.dirname(dest), exist_ok=True)
             shutil.copyfile(src, dest)
```

rprinz08, 2022-04-10:
So the problem is that specific platform `build` methods (e.g. in `build/xilinx/vivado.py`) first change working directory to build directory and then call generic_platform `copy_sources` method. `copy_sources` then copies source files into build directory by again prepending `build_dir` to destination path which leads to `build/build/...` destination path. To solve this either perform `copy_sources` in platform build method before changing working directory to `build_dir` or don't prepend `build_dir` to destination path in `copy_sources` method as proposed by @smunaut .