m-labs/migen #216: bitwise invert of unsigned int is negative

BrettRD, 2020-08-25:
inverting an unsigned integer seems to infer the existence of a sign bit, which gets set.
In this example, ~3 should be 0, but the value -4 is used to test equality.

```
from migen import *

class InvertEquality(Module):
    def __init__(self):
        self.a=Signal(2)
        self.b=Signal(2)
        self.c=Signal(1)
        self.comb += self.c.eq(self.a == ~self.b)

def test_invert_equality(dut):
    len_a, sign_a = value_bits_sign(dut.a)
    len_b, sign_b = value_bits_sign(dut.b)
    assert sign_a == False
    assert sign_b == False
    assert len_a == 2
    assert len_b == 2

    yield dut.a.eq(0)
    yield dut.b.eq(3)
    yield
    print('bitwise inverse of unsigned 2bit integer 3 is ' + str( (yield ~dut.b)))
    # bitwise inverse of unsigned 2bit integer 3 is -4
    assert (yield dut.c) == 1   # assert fails

if __name__ == "__main__":
    dut = InvertEquality()
    run_simulation(dut, test_invert_equality(dut))

```

sbourdeauducq, 2020-08-25:
This is what Python does:
```
> python
Python 3.7.6 (default, Dec 18 2019, 19:23:55) 
[GCC 9.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> ~3
-4
```

BrettRD, 2020-08-26:
Wow! i did not expect that.
Python doesn't support unsigned types or fixed bit widths, but I get the idea that the migen should follow python arithmetic conventions where possible.

I would expect the result of a bitwise operation on a fixed-size type to fit inside the original type, or generate a warning for overflow or size mismatch during comparison.

```
Python 3.6.8 (default, Aug 20 2019, 17:12:48) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> ~3 & 0x03
0
```

adding the following to the test above suggests that the inverse of 3 is still two bits wide, and still unsigned, despite being -4.
```
    len_nb, sign_nb = value_bits_sign(~dut.b)
    assert sign_nb == False
    assert len_nb == 2
```

sbourdeauducq, 2020-08-26:
> I would expect the result of a bitwise operation on a fixed-size type to fit inside the original type, or generate a warning for overflow or size mismatch during comparison.

Then you couldn't do e.g. ``Cat(signal, carry).eq(signal + 1)``, so, no.

> adding the following to the test above suggests that the inverse of 3 is still two bits wide, and still unsigned, despite being -4.

This one is a bug.

BrettRD, 2020-08-26:
I think I get it now, container properties are only enforced when they are put into named signal
Underflow in an unsigned int is handled on load into the unsigned container, not when when the subtraction takes place.

This means that for unsigned integers, the stored result of an invert operator is **never** equal to the naked expression.
As such, My gripe is with the == operator not doing anything clever with container sizes.

```
from migen import *

class SignProperties(Module):
    def __init__(self):
        self.a = Signal(2) # set to zero
        self.not_a_3 = Signal(3)
        self.not_a_2 = Signal(2)
        self.not_a_3_is_not_a = Signal(1)
        self.not_a_2_is_not_a = Signal(1)
        self.unsigned_subtraction_is_signed = Signal(1)

        self.comb += self.unsigned_subtraction_is_signed.eq((self.a-1) == -1)

        self.comb += self.not_a_3.eq(~self.a)
        self.comb += self.not_a_3_is_not_a.eq(~self.a == self.not_a_3)

        self.comb += self.not_a_2.eq(~self.a)
        self.comb += self.not_a_2_is_not_a.eq(~self.a == self.not_a_2)

def test_sign_properties(dut):
    yield dut.a.eq(0)
    yield
    # naked expressions follow signed arithmetic conventions
    assert (yield (dut.a - 1)) == -1
    # sign bits implicitly exist outside the container, and form upper bits for the larger unsigned container
    assert (yield dut.not_a_3) == 7
    # fixed size container truncates extra bits
    assert (yield dut.not_a_2) == 3
    # ~a is uses python arithmetic for signed integers of unbound size
    assert (yield dut.unsigned_subtraction_is_signed) == True

    # the stored result of a invert operator is never equal to the naked expression.
    for i in range(4):
        yield dut.a.eq(i)
        yield
        assert (yield dut.not_a_3_is_not_a) == False
        assert (yield dut.not_a_2_is_not_a) == False

if __name__ == "__main__":
    dut = SignProperties()
    run_simulation(dut, test_sign_properties(dut))
```

closed 2020-09-04 by BrettRD