m-labs/migen #219: Question: How to write tests and convert higher-level modules with other modules

dangorogoro, 2020-09-22:
I write following simple code with **Instances** to test higher-level modules with other modules.
```python
from functools import reduce
from operator import add

from migen import *
from migen.fhdl import verilog

class MyAdder(Module):
    def __init__(self, width):
        self.a = a = Signal(width)
        self.b = b = Signal(width)
        self.result =  result = Signal(width+1)
 
        self.comb += result.eq(a + b)

class MyTop(Module):
    def __init__(self):
        self.inp = Signal(8)
        self.res = Signal(9)
        adder = Instance("MyAdder",
            i_master_clk = ClockSignal(),
            i_master_rst = ResetSignal(),
            i_a = self.inp,
            o_result = self.res,
            i_b = 0x03
        )
        self.specials += adder
def adder_test(dut):    
    for i in range(20):
        yield dut.inp.eq(i)
        yield
        print("{} {}".format(i, (yield dut.res)))

dut = MyTop()
#run_simulation(dut, adder_test(dut), vcd_name="adder.vcd")

dut = MyTop()
verilog.convert(dut, ios={dut.inp, dut.res}, name="MyTop").write("out/MyTop.v")
```
It generates following verilog code and it looks very well.
```verilog
/* Machine-generated using Migen */
module MyTop(
	input [7:0] inp,
	output [8:0] res,
	input sys_clk,
	input sys_rst
);

MyAddr MyAddr(
	.a(inp),
	.b(2'd3),
	.master_clk(sys_clk),
	.master_rst(sys_rst),
	.result(res)
);

endmodule
```
Then, I tested MyTop modules with **adder_test** but I get following error.
```
raise ValueError("Could not lower all specials", self.fragment.specials)
ValueError: ('Could not lower all specials',)
```
I also use **Submodules** to test with following code.
```python
from functools import reduce
from operator import add

from migen import *
from migen.fhdl import verilog

class MyAdder(Module):
    def __init__(self, width):
        self.a = a = Signal(width)
        self.b = b = Signal(width)
        self.result =  result = Signal(width+1)
 
        self.comb += result.eq(a + b)
 
class MyTop(Module):
    def __init__(self):
        self.inp = inp = Signal(8)
        #declare the module
        self.adder = MyAdder(8)
        #add the module to our submodules list
        self.submodules += self.adder
        #assign this res is the same as the one of the submodules
        self.res = self.adder.result
        #the adder takes our input and add 3 
        self.comb += [
            self.adder.a.eq(inp),
            self.adder.b.eq(3)
        ]
def submodule_test(dut):
    for i in range(20):
        yield dut.inp.eq(i)
        yield
        print("{} {}".format(i, (yield dut.res)))
 
dut = MyTop()
run_simulation(dut, submodule_test(dut), vcd_name="out/adder.vcd")
dut = MyTop()
#verilog.convert(dut, ios={dut.inp}, name="mytop").write("out/mytop.v")
```
In this case, I could run tests for **MyTop** but the verilog code converted from **verilog.convert** has the contents of MyAdder in it. In other words, it's not separated as a module like following code.

```verilog
/* Machine-generated using Migen */
module mytop(
	input [7:0] inp
);

wire [7:0] a;
wire [7:0] b;
wire [8:0] result;

// synthesis translate_off
reg dummy_s;
initial dummy_s <= 1'd0;
// synthesis translate_on

assign a = inp;
assign b = 2'd3;
assign result = (a + b);

endmodule
```
What kind of code should I write to generate verilog code for the higher-level modules which separated from lower-level modules and to make the tests work?

sbourdeauducq, 2020-09-23:
> Then, I tested MyTop modules with **adder_test** but I get following error.

Instances are not supported in simulation. The reason for this is instances could be Verilog or VHDL, and the simulator only supports Migen.

I don't think the best way to solve this is to special-case Migen instances by adding some kind of attribute that links the instance to a Migen module so that the simulator knows what to do.
It would be better to modify the Migen Verilog generator to support breaking the output into separate files. Due in particular to various Verilog gotchas, this is not an easy task.

sbourdeauducq, 2020-09-23:
[nMigen](https://github.com/m-labs/nmigen/) supports it (by piggybacking on Yosys work to deal with the gotchas), but has a number of other issues, and I don't have time to deal with any of this at the moment.

dangorogoro, 2020-09-24:
Okay. I''ll try to see if I can do it using **nMigen**. Thank you very much! :+1:

closed 2020-09-24 by dangorogoro