m-labs/migen #259: tests fail when building with Python 3.11

gsomlo, 2022-04-03:
Trying to build migen using Python 3.11 (on Fedora rawhide, FWIW). Builds OK, but then `test_local_cd` fails during `setup.py test`.

Hopefully it's an easy fix :) -- Thanks!

```
[...]

test_local_cd (migen.test.test_examples.TestExamplesBasic) ... Traceback (most recent call last):
  File ".../examples/basic/local_cd.py", line 18, in <module>
    mm = MultiMod()
         ^^^^^^^^^^
  File ".../examples/basic/local_cd.py", line 14, in __init__
    self.submodules.foo = CDM()
                          ^^^^^
  File ".../examples/basic/local_cd.py", line 9, in __init__
    self.clock_domains.cd_sys = ClockDomain(reset_less=True)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File ".../migen/fhdl/structure.py", line 710, in __init__
    raise ValueError("Cannot extract clock domain name from code, need to specify.")

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: Cannot extract clock domain name from code, need to specify.
ERROR

[...]

======================================================================
ERROR: test_local_cd (migen.test.test_examples.TestExamplesBasic)
----------------------------------------------------------------------
Traceback (most recent call last):
  File ".../migen/test/test_examples.py", line 13, in test_method
    subprocess.check_call(
    ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

[...]
```

sbourdeauducq, 2022-04-04:
Probably needs an update of the part that reads the Python bitcode to automatically name objects.
https://github.com/m-labs/migen/blob/master/migen/fhdl/tracer.py
Are you able to take a look?

gsomlo, 2022-04-04:
On Sun, Apr 03, 2022 at 05:26:16PM -0700, Sébastien Bourdeauducq wrote:
> https://github.com/m-labs/migen/blob/master/migen/fhdl/tracer.py
> Are you able to take a look?

I'm afraid that's a bit beyond my current understanding of python inspect...

They mention some changes to the frame struct in 3.11:

https://docs.python.org/3.11/whatsnew/3.11.html

but I'm not exactly sure if referencing e.g. f_back and the like from
python is still OK or not, or how would one go about rewriting anything
that relies on directly accessing the fields of a frame...

musicinmybrain, 2022-04-08:
I looked into this a little bit. From [What’s New In Python 3.11](https://docs.python.org/3.11/whatsnew/3.11.html#cpython-bytecode-changes):

- Replaced the three call instructions: `CALL_FUNCTION`, `CALL_FUNCTION_KW` and `CALL_METHOD` with [PUSH_NULL](https://docs.python.org/3.11/library/dis.html#opcode-PUSH_NULL), [PRECALL](https://docs.python.org/3.11/library/dis.html#opcode-PRECALL), [CALL](https://docs.python.org/3.11/library/dis.html#opcode-CALL), and [KW_NAMES](https://docs.python.org/3.11/library/dis.html#opcode-KW_NAMES). This decouples the argument shifting for methods from the handling of keyword arguments and allows better specialization of calls.

The `_call_opcodes` dictionary in `migen.fhdl.tracer` is already conditionalized to handle Python version differences, but this is a somewhat bigger restructuring. I haven’t studied it enough to know how to handle it.

In the failing test/example, the opcode is `CALL` on Python 3.11.

gsomlo, 2022-11-01:
closing since merged PR fixes the problem -- thanks!

closed 2022-11-01 by gsomlo